FreeCAD/freecad.ship #5: Ship instance grayed out

joaoantoniocardoso, 2019-04-14:
Hi,

I'm trying to use freecad.ship but when trying to follow the tutorial using any of the hull from examples, the Ship instance after perform "Create a Ship Instance on top of the hull geometry" button, the ship instance was created but it is grayed out. I think it was not correctly created or it isn't recognized as it should.

Any ideas?

Thanks

---

OS: Arch Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16093 (Git)
Build type: Release
Branch: makepkg
Hash: 690774c0effe4fd7b8d2b5e2fb2b8c8d145e21ce
Python version: 3.7.2
Qt version: 5.12.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

looooo, 2019-04-14:
is there any error reported in the report-view?

joaoantoniocardoso, 2019-04-14:
@looooo thanks, I'm new using FreeCad and didn't knew the existence of the report-view, below its output:

`A3 not available
During initialization the error module 'CfdTools' has no attribute 'getModulePath' occurred in /home/joaoantoniocardoso/.FreeCAD/Mod/Cfd/InitGui.py
Please look into the log file for further information
During initialization the error invalid syntax (CADExchangerIO.py, line 131) occurred in /home/joaoantoniocardoso/.FreeCAD/Mod/CADExchanger/InitGui.py
Please look into the log file for further information
/usr/lib/python3.7/site-packages/matplotlib/__init__.py:846: MatplotlibDeprecationWarning: 
The backend.qt4 rcparam was deprecated in Matplotlib 2.2 and will be removed in 3.1. In order to force the use of a specific Qt binding, either import that binding first, or set the QT_API environment variable.
  "2.2", name=key, obj_type="rcparam", addendum=addendum)
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Ship/shipCreateShip/TaskPanel.py", line 53, in accept
    Locale.fromString(form.draft.text()))
  File "/usr/lib/freecad/Mod/Ship/shipCreateShip/Tools.py", line 57, in createShip
    ship = Instance.Ship(obj, solids)
  File "/usr/lib/freecad/Mod/Ship/Instance.py", line 44, in __init__
    tooltip = unicode(QtGui.QApplication.translate(
<class 'NameError'>: name 'unicode' is not defined`

looooo, 2019-04-15:
Seems like there are several errors involved.
1. You are using the build in Ship module and not the one from this repo. The internal module was never fully ported to python3.
2. Some error with matplotlib: maybe an incompatible version is in use (qt4)

joaoantoniocardoso, 2019-04-15:
Err... Ok, this repo is not the one installed from 0.18? It is not this repo at all or you mean that that uses an old commit?

I don't know about Qt, the default qt version on my system is qt5...

I create a fork to contribute, but I stuck in the part of use this repo as a module... How should I do? Can I edit some config of freecad to import this module?

Thanks

looooo, 2019-04-15:
If you build FreeCAD yourself you can disable the build in ship-module by theses cmake settings:
https://github.com/conda-forge/freecad-feedstock/blob/2ea4c6a81a6449b6880f8a757d0298dd889e0630/recipe/build.sh#L23L24

If you have installed freecad with a package-manager, you need to somehow disable the module. The simplest way is maybe renaming the ..../Mod/Ship/InitGui.py .../Mod/Ship/Init.py to something slightly different (eg.: .../Mod/Ship/InitGui.py.temp). Same must be done for the Plot module.

I am not yet sure if going with a seperated/external ship-module is the right way, so there are still both options available. So this is a bit messy currently. Hopefully this will change in the 0.19 development cycle...

The appimage based on conda use the external ship module (this repo) for quite some time. So this would be another option for you. I think it's also possible to develop the ship-workbench (this repo) by using the appimage. Installing the ship-workbench via the internal pip-tools (FreeCAD) in developer mode. pip-tools are already installed in the appimage

I hope this isn't too confusing and helps a bit to understand why things are a bit messy currently. But I don't want to discourage you to work on this repo. Would be nice if someone would care about the ship-tools. There are for sure some interesting things to explore and a lot of stuff to improve.

ps. after testing this myself, it seems like using the appimage is also no option as there freecad.ship and freecad.plot are isntalled by default. I guess it's best to remove this for the next iteration of the conda-based-appimages.

joaoantoniocardoso, 2019-04-24:
@looooo Thank you for your reply, Last week I managed to built it from the sources using [this script](https://gist.github.com/joaoantoniocardoso/8c4308e92455d6624eab29e694fd6e68).

Now how do I make this repository to be loaded as a module for my Freecad?

sanguinariojoe, 2021-06-21:
This issue can be closed, right?

joaoantoniocardoso, 2021-06-21:
Yeah, sorry for not closing earlier.

closed 2021-06-21 by joaoantoniocardoso