FreeCAD/freecad.ship #7: Does freecad.ship rely on freecad.plot to be installed?

luzpaz, 2019-08-30:
When loading FreeCAD.ship the terminal outputs:  
`freecad.plot is disabled, tools cannot graph output curves`

If this is the case, should the Addon Manager potentially report this and ideally ask the user if they want to install it?

looooo, 2019-08-30:
I think it's enough to give a little hint:
https://github.com/FreeCAD/freecad.ship/commit/c176f1f342df238b6daafe8322b1bdeea852eeff

What do you think?

luzpaz, 2019-08-30:
I think this is a workaround for now AND I'll make a note of this in:
* https://github.com/FreeCAD/FreeCAD-addons/issues/101
* https://github.com/FreeCAD/FreeCAD-addons/issues/29   
Just because sometimes the console isn't open for certain users or they don't pay attention to it. Eventually we're going to need to get a little more sophisticated with 3rd party dependencies. But for now your workaround is fine. Cheers!

sanguinariojoe, 2021-06-21:
This issue can be closed, right?

closed 2021-07-10 by luzpaz