FreeCAD/freecad.ship #8: Question: why there still Ship code in FreeCAD master src/Mod/Ship/*

luzpaz, 2019-09-28:
is it because it's c++ code that needs to be compiled ?

vocx-fc, 2019-09-28:
There doesn't seem to be any C++ code in the Ship workbench. If it is there still, it's because Werner and Yorik haven't made the decision to pull the plug and remove it.

If this is done finally, then some `CMakeLists.txt` files should be edited to remove the "compilation" of the workbench. In this case, compilation is just copying the files to the appropriate directories.

luzpaz, 2019-09-28:
@vocx-fc https://github.com/FreeCAD/FreeCAD/tree/master/src/Mod/Ship

vocx-fc, 2019-09-28:
I mean, the latest commit was an update to the translations, and that's fine, but if the workbench is unmaintained, and nobody really uses it, how useful are those translations?

The problem is that if the workbench becomes external, installable from the Addon Manager, it seems the translations don't quite work: [Translating external workbenches](https://forum.freecadweb.org/viewtopic.php?p=324687#p324687).

luzpaz, 2019-09-28:
Good points, @vocx-fc, this [Translating external workbenches](https://forum.freecadweb.org/viewtopic.php?p=324687#p324687) issue is coming up more frequently than ever and is something that certainly needs to be tackled very soon (I would hope say before the end of this release cycle).

vocx-fc, 2019-09-28:
To clarify my position, as Ship and Plot can be installed from the Addon Manager, I don't mind if both of them are removed from the master tree, even if this breaks their translations. I think the translation issue is something that can be solved later.

Both workbenches are already turned off by default in the standard CMake configuration, so removing them from the source removes some clutter. Otherwise, the same code is duplicated in the main tree, and in their own repository.

luzpaz, 2019-09-28:
We should probably move the current translated strings from Crowdin for both Ship and Plot in to their respective github repos before we remove their translation files from master. I say this to preserve the work translators have done after we made them external workbenches.

vocx-fc, 2019-09-28:
Yes, of course.

sanguinariojoe, 2021-06-21:
This issue can be closed, right?

luzpaz, 2021-07-10:
@yorikvanhavre said he would work on this.

sanguinariojoe, 2021-07-10:
It is about to be removed:

https://github.com/FreeCAD/FreeCAD/pull/4900

closed 2021-07-25 by luzpaz