FreeCAD/freecad.ship #15: Roadmap for this Workbench

luzpaz, 2021-07-10:
Let's talk about the roadmap. What are some of the enhancements/features to be proposed?

sanguinariojoe, 2021-07-11:
Well, getting everything in place again has total priority:

 - Removing Ship (workbench and module) from core (https://github.com/FreeCAD/FreeCAD/pull/4900)
 - Removing Plot workbench from core
 - Enabling Plot module by default on core
 - Removing outlines draw tool from this workbench (which is nicely covered by the default tools)
 - Upgrading documentation

luzpaz, 2021-08-06:
>     * Removing Ship (workbench and module) from core ([Ship: Remove Ship module FreeCAD#4900](https://github.com/FreeCAD/FreeCAD/pull/4900))
>     * Removing Plot workbench from core
>     * Enabling Plot module by default on core

Waiting on @yorikvanhavre and @wwmayer to approve these

sanguinariojoe, 2021-08-16:
Ship removed! 4 to go!

luzpaz, 2021-08-16:
@sanguinariojoe awesome. Question, are the developments in this workbench still pending because the other issue in the roadmap aren't officially resolved?

sanguinariojoe, 2021-08-16:
Yes! It would result in terrible conflicts trying to solve them altogether. After solving point 3 everything will flow

yorikvanhavre, 2021-08-17:
let's go after it! :)

luzpaz, 2021-09-07:
ping

sanguinariojoe, 2021-10-04:
Karnotcycle is taking care on the outlines draw tool (https://github.com/FreeCAD/freecad.ship/pull/22).

I was a bit busy fixing https://github.com/FreeCAD/freecad.ship/issues/21, and now I hope to come back to the integration of https://github.com/mancellin/capytaine

Meanwhile/afterwards I am addressing https://github.com/FreeCAD/freecad.ship/issues/17, where I am probably adding info about the roadmap