FreeCAD/freecad.ship #16: Feature: Integration workflow with Free!Ship

luzpaz, 2021-07-25:
> FREE!ship is a surface modeling program for designing ships yachts and boats. Subdivision surfaces are used rather than NURBS to give the designer the freedom to design ANY desired hullshape. Unfold surfaces, panel based hydrostatics, various exp formats

https://sourceforge.net/projects/freeship/

lmasellis, 2021-08-21:
Hello,

Seems like development of the original software on Sourceforge has been abandoned long time ago.

A more recently updated fork is available here:

https://github.com/markmal/freeship-plus-in-lazarus

sanguinariojoe, 2021-10-14:
I wanted to reply to this issue for a long time. It would be nice to get some integration with FreeShip!, but I really doubt it will be happening.

First of all, FreeCAD-Ship and FreeShip! are based in completely different theories. While FreeShip! is based on surfaces (and sections for some tools), FreeCAD-Ship is based on solids. I am not addressing here the benefits and drawbacks of each approach.

Secondly, FreeCAD-Ship offers the same tools available in FreeShip! (except advance resistance regressions, which I am not really sure I want to add). Moreover, FreeCAD already offer things that FreeShip! is missing, like weights and tanks definition. And it will offer much more things soon, like seakeeping tools.

Finally, FreeShip! is dying (if not already dead), so I really think it would be a waste of time.

I am waiting for some feedback before closing this issue

closed 2022-12-26 by sanguinariojoe