FreeCAD/freecad.ship #17: README: elaborate list of features that already exist in the Workbench

luzpaz, 2021-07-25:
Lets help users understand what is currently possible in the Ship workbench on the README when they read about it in the Addon Manager

luzpaz, 2021-08-06:
Uploading images from wiki to github to feature in README.md

[image]

[image]

[image]

[image]

[image]

sanguinariojoe, 2021-08-16:
It automatically closed...

sanguinariojoe, 2021-10-14:
[Here we go!](https://github.com/FreeCAD/freecad.ship/blob/master/README.md)

closed 2021-10-14 by sanguinariojoe

luzpaz, 2021-10-15:
Like a boss! :sunglasses:

commit ba13a7a8 (2021-08-16): Updated README

Closes #17
Added missing info + extra content