FreeCAD/freecad.ship #21: Feature request: ability to visualize waterline, CoB and CoG

martijn-heil, 2021-09-29:
As per title, I think a useful feature for hull design would be the ability to visualize the actual waterline, center of buoyancy and center of gravity of a ship given the weights defined in this workbench.

sanguinariojoe, 2021-10-01:
I am coming with something soon!

martijn-heil, 2021-10-01:
As an additional note I think it would be very useful if you could do this with angles of heel too, so for example you can visualize the waterline of a heeled-over sailboat. Especially with modern hull design trends with hard chines and wide beam this is important to see when choosing their placement on the hull.

sanguinariojoe, 2021-10-04:
Hey Martijn!

Check this out: https://github.com/FreeCAD/freecad.ship/pull/23

I added a new tool called Sink and trim, which is computing the equilibrium point on top of a load condition. You can show the results in the ship upright position, or using the free-surface as the reference (i.e. the free surface is the z=0 plane).

I am merging it right now, so you can test it with the addon manager. Let me know!

P.S. Some changes I made might broke some other tools, so if you test all the tools I will really appreciate that

martijn-heil, 2021-10-04:
Thanks for the quick work, I'll take a look!

martijn-heil, 2021-10-04:
Maybe I'm just doing something stupid, but it looks as if the COG of the hull isn't calculated correctly.
I've got a very simplistic scow-bowed hull shape that I'm trying it on, with a cut out in the back that's meant to resemble some kind of large open cockpit. 
I've defined a single weight (the assumed weight of the structure) at the default location it places the weight (the centre).
Subsequently I've executed the Sink and trim tool with 'Reference on the free surface', which leaves me with this result;
[image]
Simply looking at it by eye, it seems like the COG is too far aft.
Further investigation by calculating and illustrating the actual COG of the resulting `SinkAndTrim_Ship` body using the FCInfo macro reveals the actual COG at the centre of the crosses:
[image]
Which appears to confirm that indeed the COG calculated by Sink and trim is too far aft.

This could  explain the rather unnatural looking trim that my simplistic hull shape has attained.
[image]

Location of the single weight *(note that it is in front of the cockpit)
(it may seem a little off-center in this screenshot but I can assure you that it is in fact located at the transversal center)*:
[image]

Some more pictures to clarify further the hull shape:
[image]
[image]
[image]

sanguinariojoe, 2021-10-05:
github automatically closed the issue...

So, as far as I understood you have created the weight selecting the ship itself and executing the tool. Thus you expect your COG is in the COG of your volume, right?

Confirm me that so I start debugging it (you may also attach your FreeCAD project if you want)

martijn-heil, 2021-10-05:
> So, as far as I understood you have created the weight selecting the ship itself and executing the tool. Thus you expect your COG is in the COG of your volume, right?

That is correct.
Here's my project if you want to take a look:
[project.zip](https://github.com/FreeCAD/freecad.ship/files/7284704/project.zip)

martijn-heil, 2021-10-05:
I also have a suspicion that the entered weight in kilogram is treated as a much heavier weight off by a power of 10, like 100 kg is treated as 1000 kg. This needs to be tested further to be confirmed though.

Furthermore, after doing a Sink and trim the results also include some text which contains:
`Δ = 1.80 t`
Maybe it is a naval architecture thing, I'm a programmer, not a properly trained naval architect, but what does it mean?
It seems to me that it's the displacement, but that doesn't make sense because I get this displacement with a single weight of 100 kg (100 kg!, not 1000!). 
When I measure the actual displacement of the underwaterbody again with FcInfo it is indeed very close to 1.8 metric tons.
Is there some kind of hidden intrinsic weight attached to the ship that I can't see anywhere?

On the wiki in the second part of the Freecad ship tutorial it says:
> Weights definition tool can be used to set first category of weights. When you launch the tool for first time (with ship instance selected), the Ship workbench initializes ship weights with Lightweight ship (equal to ship displacement) that is placed on the ship geometry center of gravity X coordinate, and at design draft height.

I haven't seen this behaviour, I guess the wiki tutorial is here slightly out of date?

sanguinariojoe, 2021-10-05:
Ok, I will take a look, don't worry!

> I also have a suspicion that the entered weight in kilogram is treated as a much heavier weight off by a power of 10, like 100 kg is treated as 1000 kg. This needs to be tested further to be confirmed though.

It sounds like a gravity doing funny things.

> but what does it mean?

Yes, Δ is the displacement, which yes, shall be equal to the weight

> On the wiki in the second part of the Freecad ship tutorial it says:

Yeah, the wiki is outdated, forget about it for the time being. See https://github.com/FreeCAD/freecad.ship/issues/15

martijn-heil, 2021-10-05:
As a side note, I haven't calculated it by hand so far, but simply looking at it by eye I think the proper natural trim of this simplistic hull shape will be bow-down, considering the lack of buoyancy at the bow and the lack of weight due to the cockpit cutout in the back. I definitely doubt that it will have 0 degrees trim, which is why it makes a nice test case for this feature.

sanguinariojoe, 2021-10-07:
Well, I am pretty sure this is a problem of lack of documentation rather than a bug...

I have downloaded your project, removed the load condition as well as the weight, selected the ship, executed the weight creation tool, set a **density** of 300 kg/m3, and executed the sink and trim tool:

[image]

I suspect you created your weight, and then selected and edited it in the data tab, removing the density and setting the mass. Am I right? Another possibility is that you just simply set the quantity in kg instead of kg/m3. Whatever way you did so, you are instructing FreeCAD-Ship to consider a punctual weight, so it will compute the COG as the average of the shape vertices. Actually the user should never manipulate the weights in the data tab. 

Let me know if you hacked the weight on the data panel or you did that in the input box please

sanguinariojoe, 2021-10-07:
> One thing I would like to add though, everytime I immediately change the opacity of the generated free surface to 0, so that it isn't seethrough. This makes the visualization a lot clearer in my opinion.

Ok! I am making it opaque in the next commit.

Feel free to open another issue about the weights creation and the units

sanguinariojoe, 2021-10-14:
Hey @martijn-heil , as you can see I have changed the way the user may introduce inputs, in such a way different units can be used, but they should be compatible with the expected value. This way, it is not possible to set the weight in kg for volumes anymore.

I suppose that is finally fixing this issue! Please, check and feel free to open new issues if you find more errors

commit a437c791 (2021-10-04): Merge pull request #23 from sanguinariojoe/equilibrium

Fixes #21
commit bc503f84 (2021-10-04): [Fixes #21] Sink and trim tool
closed 2021-10-14 by sanguinariojoe