FreeCAD/freecad.ship #33: Circular import when adding weights

albertoferna, 2021-11-15:
Hello all,

This workbench is looking increasingly interesting. I've been doing some test with it and I've found what I believe is a bug.
I'm running ubuntu 20.04 and both the stable appimage, and the weekly one. The issue appears on both.
The way to reproduce it is fairly simple. After opening freecad and activate the ship module I load the example (s60_IowaUniversity). Then, with ship selected, I click on create a new ship weight (I tried both icon and menu item). Immediately I get the following error:

```
12:43:36  Running the Python command 'Ship_Weight' failed:
Traceback (most recent call last):
  File "/home/user/.FreeCAD/Mod/Ship/freecad/ship/ShipGui.py", line 119, in Activated
    from . import shipCreateWeight
  File "/home/user/.FreeCAD/Mod/Ship/freecad/ship/shipCreateWeight/__init__.py", line 23, in <module>
    from .TaskPanel import createTask
  File "/home/user/.FreeCAD/Mod/Ship/freecad/ship/shipCreateWeight/TaskPanel.py", line 27, in <module>
    from . import Tools
  File "/home/user/.FreeCAD/Mod/Ship/freecad/ship/shipCreateWeight/Tools.py", line 26, in <module>
    from .. import WeightInstance as Instance
  File "/home/user/.FreeCAD/Mod/Ship/freecad/ship/WeightInstance.py", line 31, in <module>
    from .shipCreateWeight.Tools import compute_inertia

cannot import name 'compute_inertia' from partially initialized module 'freecad.ship.shipCreateWeight.Tools' (most likely due to a circular import) (/home/user/.FreeCAD/Mod/Ship/freecad/ship/shipCreateWeight/Tools.py)
```

I've tried with different shapes just to make sure it is not some geometry issue. 
The versions I tried are:

- FreeCAD_weekly-builds-26306-Linux-Conda_glibc2.12-x86_64.AppImage
- FreeCAD_0.19-24291-Linux-Conda_glibc2.12-x86_64.AppImage

albertoferna, 2021-11-15:
Just some more information.
Version FreeCAD_weekly-builds-26306-Win-Conda_vc14.x-x86_64 in windows 10 also gives the same error. The addon is installed using the add-on manager

sanguinariojoe, 2021-11-15:
Yeah, I think you found a bug... I am fixing this soon

albertoferna, 2021-11-16:
I tried your solution and it looks like it is solved.
I have some other problems but they look to be related with how robust the cutting mechanism is in OCC.
I'm going to close this issue then.

commit 6f580ba0 (2021-11-16): Fixes #33
closed 2021-11-16 by albertoferna