FreeCAD/freecad.ship #46: [FEATURE REQUEST] Add release

darigovresearch, 2022-10-23:
Would be nice to have a Vx.x.x release in GitHub to make it clearer to see when there are major & minor releases.

Instructions on how to do this can be seen [here](https://docs.github.com/en/repositories/releasing-projects-on-github/managing-releases-in-a-repository)

sanguinariojoe, 2022-10-24:
I develop this time to time, I do not think releases would fit in the chaotic development scheme I have for this.

darigovresearch, 2022-10-29:
No worries at all if it's a chaotic development scheme, it can be a v0.1.0 and would still provide guidance to others if you make a minor fix release or a major feature release.

More details can be found here - http://semver.org/