FreeCAD/freecad.ship #51: "Plots the ship hull outline draw" isn't included in last Ship Workbench Version

MMNTSD, 2022-12-06:
Hi,
Please view the last version due that the **Plots the ship hull outline draw** isn't include in the last version. (Lines drawing)
I tried with Addon Manager and the result is that the module of Lines drawing doesn't included. Also I went to the : <user name/AppData/Roaming/Mod/Ship/freecad/ship/**shipOutlineDraw** > does not exist. ****

luzpaz, 2022-12-17:
Please add your full [About](https://wiki.freecad.org/About) info

MMSN2, 2022-12-21:
Luzpaz,
Here is:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.0.0
  * Ship 2022.4.11
[/code]

luzpaz, 2022-12-21:
@sanguinariojoe ping

sanguinariojoe, 2022-12-26:
Hey guys! Sorry for taking so long to reply!

Outlines draw was marked as outdated back on FreeCAD 0.19, and dropped since FreeCAD 0.20. That is because FreeCAD itself already offer the tools to do that

closed 2022-12-26 by sanguinariojoe

MMSN2, 2022-12-27:
Hello Jose,I think the Out Draw  is so necessary due that the Waterline and Buttock input give a good parameter definition.Thanks in advance,
Giovanni