FreeCAD/freecad.ship #59: The tutorial mentions a lines drawing tool, but I don't see it?

tim-rohrer, 2023-12-16:
As the title says...

I'm experimenting with this add on, but I don't see the Lines Drawing tool in the workbench.

Does this tool exist any more? Searched current code for expected terms, but no luck.

sanguinariojoe, 2023-12-16:
Yeah, sorry, it us outdated. The outlines draw was dropped because freecad
offers tools to do so out of the box

tim-rohrer, 2023-12-18:
Thanks for the quick reply. 

Can you give me a hint (or point to a tutorial/article) that describes the FreeCAD tools for doing this?

Thanks!

shaise, 2023-12-18:
MangoJelly have some nice tutorials:
https://www.youtube.com/watch?v=-539fgU50Bw

sanguinariojoe, 2023-12-18:
To make the slices:

https://wiki.freecad.org/Part_CrossSections

To plot:

https://wiki.freecad.org/TechDraw_Workbench

tim-rohrer, 2023-12-18:
That should be enough to get me started. Thank you!

closed 2023-12-18 by tim-rohrer