FreeCAD/freecad.ship #60: Crash when showing hydrostatics graph

facorazza, 2024-08-31:
When trying to compute hydrostatics for the series 60 example FreeCAD crashes upon visualizing the graph.

I'm using FreeCAD 0.21.2

sanguinariojoe, 2024-09-06:
Weird, noone else complained... I will check this out asap.

Meanwhile would you post your OS details?

facorazza, 2024-09-06:
> [Tracking image][https://github.com/notifications/beacon/AE42KEMVC37IZLQFIBEEPKTZVFYV5A5CNFSM6AAAAABNOFM6UOWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTULDCS5A.gif]

I'm on Ubuntu 22.04.4 LTS

Thanks

luzpaz, 2024-09-06:
@facorazza please add your full [About](https://wiki.freecad.org/About) info. TIA

sanguinariojoe, 2024-09-12:
Everything works just fine to me. My system info:

```
OS: EndeavourOS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.5, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * lattice2 1.0.0
  * workfeature
  * Ship 2022.4.11
```

[image]

Anyway, after some changes where required for translations, I upgraded the system resources... It might happens that the former ship workbench was trying to make some sort of soup of imports between PySide2 and PySide6.

Would you give it a shot again?

Also, would you send us the log? [Check this out](https://wiki.freecad.org/Debugging)

facorazza, 2024-09-14:
> @facorazza please add your full [About](https://wiki.freecad.org/About) info. TIA

The reply by email didn't work for some reason.

```
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.28
  * Render 2024.8.4
  * Cfd
  * Ship 2022.4.11
  * CfdOF 1.27.7
  * CurvedShapes 1.0.9
  * Plot 2022.4.17
  * woodworking 0.21.2.33771
```

and here's the log:

```
Computing transversal areas...
Some Inventor representation errors can be shown, please ignore them.
1 / 28
2 / 28
3 / 28
4 / 28
5 / 28
6 / 28
7 / 28
8 / 28
9 / 28
10 / 28
11 / 28
12 / 28
13 / 28
14 / 28
15 / 28
16 / 28
17 / 28
18 / 28
19 / 28
20 / 28
21 / 28
22 / 28
23 / 28
24 / 28
25 / 28
26 / 28
27 / 28
28 / 28
Done!
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x3ee80) [0x75a4c1651e80]
#1  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x132560) [0x75a4c3332560]
#2  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x15ab09) [0x75a4c335ab09]
#3  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyType_GenericAlloc+0x15) [0x75a4c335abc5]
#4  /app/lib/python3.11/site-packages/numpy/core/_multiarray_umath.cpython-311-x86_64-linux-gnu.so(+0x1e4e14) [0x75a4525e4e14]
#5  /app/lib/python3.11/site-packages/numpy/core/_multiarray_umath.cpython-311-x86_64-linux-gnu.so(+0x1e58f8) [0x75a4525e58f8]
#6  /app/lib/python3.11/site-packages/numpy/core/_multiarray_umath.cpython-311-x86_64-linux-gnu.so(+0x292232) [0x75a452692232]
#7  /app/lib/python3.11/site-packages/numpy/core/_multiarray_umath.cpython-311-x86_64-linux-gnu.so(+0x294e2b) [0x75a452694e2b]
#8  /app/lib/python3.11/site-packages/numpy/core/_multiarray_umath.cpython-311-x86_64-linux-gnu.so(+0x2c7350) [0x75a4526c7350]
#9  /app/lib/python3.11/site-packages/numpy/core/_multiarray_umath.cpython-311-x86_64-linux-gnu.so(+0x382b25) [0x75a452782b25]
#10  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x353b) [0x75a4c330691b]
[... 41 lines cut ...]
#52  0x75a4c2bb6949 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x309
#53  0x75a4c2bb6949 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x309
#54  0x75a4c2bb6949 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x309
#55  0x75a4c2bb6949 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x309
#56  0x75a4c2bb6949 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x309
#57  0x75a4c2bb540f in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x52f
#58  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x18bfb1) [0x75a4c2b8bfb1]
#59  0x75a4c2bbd326 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x8f6
#60  0x75a4c2b7ad6e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#61  0x75a4c4a55180 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
#62  0x75a4c1edfca8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#63  0x75a4c1ee32a1 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x191
#64  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x339e93) [0x75a4c1f39e93]
#65  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x239) [0x75a4c0117e59]
#66  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xbe2b8) [0x75a4c01752b8]
#67  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x75a4c0115513]
#68  0x75a4c1f39586 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x66
#69  0x75a4c1ede73b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#70  0x75a4c1ee6bd0 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x90
#71  0x75a4c4981b25 in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0x1e75
#72  FreeCAD(+0x9387) [0x63b2414e6387]
#73  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x2808a) [0x75a4c163b08a]
#74  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x75a4c163b14b]
#75  FreeCAD(+0x96c5) [0x63b2414e66c5]
```

sanguinariojoe, 2024-09-21:
It is clear that the task finished, but I am seeing this line on the stacktrace:

`#21  0x75a4c4a55180 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90`

Theoretically, FreeCAD should be showing you the hydrostatics on runtime, while they are computed. But as far as I understand that is not happening to you. You are never seeing any plot, right? Not even an empty one.

It seems to me like your FreeCAD is trying to plot all at once at the end, and that is making FreeCAD crash.

Would you try to plot something simpler? [Look here](https://wiki.freecad.org/Plot_Basic_tutorial)

facorazza, 2024-09-21:
When I try to plot the areas curve I can see the graph for a few seconds and then FreeCAD crashes. If I try to plot the hydrostatics I don't see any graphs before the crash.

It is something wrong with the plotting though.

If I try to run:

```python
try:
    from FreeCAD.Plot import Plot
except ImportError:
    from freecad.plot import Plot

Plot.figure("TrigonometricTest")
```

I get the plot for a few seconds before FreeCAD crashs just like when plotting the areas curve.

Should I open an issue on the FreeCAD repo? I also tried with Ondsel and it works fine. Could it be the flatpak release that is problematic?

sanguinariojoe, 2024-09-21:
Yeah, please open a ticket on the FreeCAD repo.

The plotting capabilities were integrated on FreeCAD some time ago (they
are not part of the Plot module anymore). On top of that, you also noticed
that has to do with the flatpak package.

I keep this open for a while though.

facorazza, 2024-09-21:
I have created the issue [here](https://github.com/FreeCAD/FreeCAD/issues/16709)