FreeCAD/freecad.ship #65: No "Lines Drawing" icon in ship workbench

deichrenner, 2024-10-22:
```
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Ship 2022.4.11
```

The documentation lists a "Lines Drawing" functionality which is listed with the icon 
[image]
However, I'm not able to find this functionality in the workbench. What am I missing here?

hasecilu, 2024-10-23:
I don't see it on the Wiki: https://wiki.freecad.org/Ship_Workbench

luzpaz, 2024-10-23:
@deichrenner you also may want to consider migrating to RC2 since 0.21.x bugs won't be addressed by the freecad devs anymore. The addon devs may still support it but it may be best to just migrate because there are so many fixes and features in 1.0RC2 that it's worth it.